tobozo/ESP32-ENC28J60 #1: Some files are unknown

simogaspa84, 2022-10-11:
Hi @tobozo i finally have time to test this lib.. 

I am working under platformio with arduino framework..   framework = arduino

Some of the files you added in the folder src are not present..  

[image]

tobozo, 2022-10-11:
Hi,

this library is for espressif32 package 2.0.5 only, you'll have to force platformio to use it, or use Arduino IDE.

```ini
platform = espressif32 @ ^5
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.5/esp32-2.0.5.zip
```

closing this as not a bug

simogaspa84, 2022-10-11:
ok so this setting should work 

`platform = espressif32@3.5.0
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.5/esp32-2.0.5.zip
board = esp32dev
framework = arduino`

tobozo, 2022-10-11:
I've pushed a platformio example and also published the library to the platformio registry, should be easier now

simogaspa84, 2022-10-11:
Thanks @tobozo  ... I have tried in this way but still I have many errors about the missing files 

``[env:esp32dev]
platform = espressif32@3.5.0
board = esp32dev
framework = arduino
check_tool = cppcheck
check_flags = cppcheck: --enable=all --language=c++ --addon=misra.json
lib_ldf_mode = deep
lib_deps = 
	me-no-dev/AsyncTCP@^1.1.1
	me-no-dev/ESP Async WebServer@^1.2.3
	emelianov/modbus-esp8266@^4.0.0
	bblanchon/ArduinoJson@^6.19.2
	https://github.com/maxgerhardt/EthernetENC.git
	knolleary/PubSubClient@^2.8
	khoih-prog/Functional-Vlpp@~1.0.2
	chrisjoyce911/esp32FOTA@^0.2.3
	ESP32-ENC28J60

tobozo, 2022-10-11:
you get errors because you chose to ignore my suggestions of using custom `platform` and `platform_packages` values in your platformio environment.

just try this [platformio example](https://github.com/tobozo/ESP32-ENC28J60/tree/main/examples/pio_example) **as it is** to confirm you're still getting errors.

on the other hand if the platformio example works then the problem comes from your project and I'm afraid it's up to you to fix it