tobozo/ESP32-ENC28J60 #4: Save firmware

KJPOUNDY132, 2022-10-26:
Hi is it just get firmware on every reset or when it gets once use until another firmware update?
Btw its realy very great job thx for your time and work

tobozo, 2022-10-26:
Hi, thanks for your feedback :+1: 

The FOTA example is only provided as a proof of concept, all it does is download a hello world sketch and flash the ESP with it.

I suggest you take a look at the [esp32FOTA](https://github.com/chrisjoyce911/esp32FOTA) library as firmware updating is a bit outside the scope of this driver.

KJPOUNDY132, 2022-10-26:
if reset esp after flashing hello world sketch.  does it use internal  or download one more time?

tobozo, 2022-10-26:
your question means you don't know how OTA behaves on esp32, but this is too long to explain here

this video is a good start https://www.youtube.com/watch?v=zXqhv_iy_lI

closing this as not an issue

closed 2022-10-26 by tobozo