tobozo/ESP32-ENC28J60 #7: rebooting ESP32 after Connection started

prajil-p, 2022-12-05:
hi ,
I'm using the  ETH_ENC28J60 example , and directly flash to esp32dev module without changing program, I'm using node MCU ESP-32s v1.1 development board. The program always restart after connection started.  also use the same 10 pin ethernet module mentioned in this document, 

Ethernet LAN module connected with esp32 according to the configuration

#define SPI_HOST       1
#define SPI_CLOCK_MHZ  8
#define INT_GPIO       4

#define MISO_GPIO 12
#define MOSI_GPIO 13
#define SCLK_GPIO 14
#define CS_GPIO   15

Serial monitor always shows like this

mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
ETH Started
Connecting...
Connecting...
ETH Connected
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (enc28j60_tsk) 
Core  1 register dump:
PC      : 0x40083e61  PS      : 0x00060f36  A0      : 0x80083f18  A1      : 0x3ffaefa0  
A2      : 0x00000054  A3      : 0x00001800  A4      : 0x00001004  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffb0874  A8      : 0x3ffaf0c0  A9      : 0x3ffaf090  
A10     : 0x3ffb0874  A11     : 0xffffffff  A12     : 0x3ffb8000  A13     : 0x00000000  
A14     : 0x3ffb6f84  A15     : 0x00000022  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008a855  LEND    : 0x4008a865  LCOUNT  : 0xfffffffb  

Backtrace:0x40083e5e:0x3ffaefa00x40083f15:0x3ffaefd0 0x40083f45:0x3ffaeff0 0x400930f1:0x3ffaf010 0x4008df61:0x3ffaf030 0x4008e21c:0x3ffaf050 0x40085038:0x3ffaf070 0x4008506a:0x3ffaf090 0x400851e1:0x3ffaf0c0 0x400e8a89:0x3ffaf0e0 0x400e7d87:0x3ffaf100 0x400e75bd:0x3ffaf120 0x4008a599:0x3ffaf140 0x40088f06:0x3ffaf160 0x40088f90:0x3ffaf180 0x4008947d:0x3ffaf1b0 0x4016102b:0x3ffaf1e0 0x4015a6f3:0x3ffaf200 0x4015a871:0x3ffaf510 0x4016c695:0x3ffaf540 0x40092e6d:0x3ffaf570 0x400dd966:0x3ffaf5c0 0x400de3a6:0x3ffaf5f0 0x400de500:0x3ffaf620 0x400d3754:0x3ffaf640 0x400d3afd:0x3ffaf6a0 0x400d391b:0x3ffaf6e0 

ELF file SHA256: 0000000000000000

Rebooting...

please help me to solve this problem
thank you

tobozo, 2022-12-05:
hi, thanks for your feedback

The [example](https://github.com/tobozo/ESP32-ENC28J60/blob/main/examples/ETH_ENC28J60/ETH_ENC28J60.ino) has been tested with an esp32-wroom with the exact same pin settings, using arduino-esp32 2.0.5.

Can you confirm that the example sketch wasn't modified, and nothing else but the ENC28J60 was connected to the ESP32 during your test?

I suggest you use the [Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder) to get more info on this stack trace.

The `Stack canary watchpoint triggered` part in the error message only hints at a memory exhaustion in the stack, and decoding the exception could tell if this happens in loop() or in another task:

Things you can try:

- Check the wiring: do you have exactly 7 short wires flying between the ethernet module and the esp32?
- Use the other spi bus: `#define SPI_HOST 0`
- Confirm the problem also happens with one of the other libraries implementing ENC28J60: 
  - https://github.com/khoih-prog/WebServer_ESP32_ENC
  - https://github.com/khoih-prog/AsyncWebServer_ESP32_ENC

tobozo, 2024-06-18:
closing this as inactive

closed 2024-06-18 by tobozo