tobozo/ESP32-ENC28J60 #8: Version of Arduino too Old

RaysoftTechnologiesInc, 2023-03-14:
Hi,

I am hoping you could give me a pointer in regards to this issue.
[image]
 running the program tells me that arduino version is too old, I have tried updaing the arduino, but did not change anything.

tobozo, 2023-03-20:
hi,

Sorry about the misleading message, it is referring to the [arduino-esp32 core](https://github.com/espressif/arduino-esp32) version, not the IDE version.

ESP32 Core version can updated from the boards manager in Arduino IDE.

RaysoftTechnologiesInc, 2023-03-20:
Hi,
Thank you for your response, but my ESP32 Core version in the Arduino is up to date, I already have this one installed.
[image]

Is there anything i am still doing wrong?

tobozo, 2023-03-20:
1.0.6 doesn't sound like up to date, current version is 2.0.7, you should probably update the `package_esp32_index.json` board url in the arduino preferences.

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

RaysoftTechnologiesInc, 2023-03-20:
I update the package and now and it seems like I am now going past thaterror, However, it seems like there are some files that not being found by when running the program.
[image]

tobozo, 2023-03-20:
looks like the old 1.0.6 is still there and interfering, did you remove 1.0.6 before installing 2.0.7 ? 

a clean install of Arduino IDE may be required, I wish I could help you with that but I have no idea how to do it on Windows

RaysoftTechnologiesInc, 2023-03-20:
Hi 

Thank you, I had to uninstall the version 1.06.

Working now.

Thank you I really appreciate.

closed 2023-03-20 by RaysoftTechnologiesInc