tobozo/ESP32-ENC28J60 #12: How to configure the mac address manualy ?

Hedmilson, 2023-09-05:
I need to assign a new  MAC number (not default) .
tanks

tobozo, 2023-09-05:
hi,

there's a method for that:

```c
ETH.macAddress(uint8_t* mac);
```

Hedmilson, 2023-09-05:
Thanks for replying, but this method is not working as expected.

How to declare the new mac?
ex: mac[] = {0x55,0x34......
and how to call the ETH.macAddress method.
Before or after ETH.config ?
Thanks again..

tobozo, 2023-09-06:
> Thanks for replying, but this method is not working as expected.

"not working" is a very general term and I'm not very good at guessing, does it fail to compile? does it crash? is your custom mac address ignored?

> How to declare the new mac?

according to espressif documentation you can declare your ethernet mac as follows:

```c
uint8_t my_eth_mac_address[6] = {0x11, 0x22, 0x33, 0x44, 0x55, 0x66};
ETH.macAddress( my_eth_mac_address );
```

But I never tested that since this driver is just a quick and dirty proof of concept.

Chances are it's not completely implemented and would inherit from a [hardcoded Mac Address](https://github.com/tobozo/ESP32-ENC28J60/blob/main/src/ESP32-ENC28J60.cpp#L90) instead.

You may have to edit the cpp file in the library folder and hardcode your mac address there.