tobozo/ESP32-ENC28J60 #16: The following error occurred after installation of 2.0.0.

topman-joung, 2024-06-18:
First of all, thank you always for the good data.
After this update, an error occurred as follows.

esp32 = 3.0.1 is currently installed
ESO32-ENC28J60 = 2.0.0 is installed.

c:\Users\ATOZSYSTEM\Documents\Arduino\libraries\ESP32-ENC28J60\src\ESP32-ENC28J60.cpp: In member function 'bool ENC28J60Class::begin(int, int, int, int, int, int, int, uint8_t*)':
c:\Users\ATOZSYSTEM\Documents\Arduino\libraries\ESP32-ENC28J60\src\ESP32-ENC28J60.cpp:104:28: error: 'enc28j60_begin' was not declared in this scope
  104 |   esp_eth_mac_t* eth_mac = enc28j60_begin(MISO_GPIO, MOSI_GPIO, SCLK_GPIO, CS_GPIO, INT_GPIO, SPI_CLOCK_MHZ, SPI_HOST);
      |                            ^~~~~~~~~~~~~~
c:\Users\ATOZSYSTEM\Documents\Arduino\libraries\ESP32-ENC28J60\src\ESP32-ENC28J60.cpp:114:28: error: 'esp_eth_phy_new_enc28j60' was not declared in this scope; did you mean 'esp_eth_phy_new_w5500'?
  114 |   esp_eth_phy_t* eth_phy = esp_eth_phy_new_enc28j60(&phy_config);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~
      |                            esp_eth_phy_new_w5500
c:\Users\ATOZSYSTEM\Documents\Arduino\libraries\ESP32-ENC28J60\src\ESP32-ENC28J60.cpp:136:7: error: 'emac_enc28j60_get_chip_info' was not declared in this scope
  136 |   if (emac_enc28j60_get_chip_info(eth_mac) < ENC28J60_REV_B5 && SPI_CLOCK_MHZ < 8) {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\ATOZSYSTEM\Documents\Arduino\libraries\ESP32-ENC28J60\src\ESP32-ENC28J60.cpp:136:46: error: 'ENC28J60_REV_B5' was not declared in this scope
  136 |   if (emac_enc28j60_get_chip_info(eth_mac) < ENC28J60_REV_B5 && SPI_CLOCK_MHZ < 8) {

tobozo, 2024-06-18:
hi, and thanks for your feedback :+1:  

I'm guessing you're not using an esp32 but one of esp32-{s2,s3,c3,c6,h2} ?

apparently espressif restricted `emac_ext_struct.h` to esp32 only 🙄, so I imported yet another file from esp-idf and added a conditional macro for all esp32 variants, but this is untested code so I'll need a confirmation that it works before producing a new release.

topman-joung, 2024-06-20:
The product I'm using is
It's LOLINS3 PRO as ESP-32-S3.
Prior to this update, it worked without any problems when using Lib for the product.
However, after this update, the same situation as the above problem occurred.
Thank you always for your help ^^

tobozo, 2024-06-20:
please try [ESP32-ENC28J60 v2.0.1](https://github.com/tobozo/ESP32-ENC28J60/releases/tag/v2.0.1)

topman-joung, 2024-06-21:
빠른 조치에 감사드립니다.
잘 실행됩니다.

commit 95f13807 (2024-06-18): added support for all esp32 family devices, fixes #16
closed 2024-06-21 by topman-joung