tobozo/ESP32-ENC28J60 #18: cant find esp_netif_types.h

Criazel, 2024-07-05:
fatal error: esp_netif_types.h: No such file or directory

tobozo, 2024-07-05:
Hi,

I noticed that you have posted an error message without any context or greeting. While I appreciate your input, it would be helpful if you could provide more information and rephrase your message to include a polite introduction and closing.

This will make it easier for me to understand and address your concern, and in a more general context shows respect for whomever assists you.

Thank you for your cooperation and understanding in this matter.

Criazel, 2024-07-05:
whoops, my bad, im kinda new to github and esp, so i was trying to compile the eth example code and i got the error of 
fatal error: esp_netif_types.h: No such file or directory, thanks

tobozo, 2024-07-05:
being new in a community has never been an excuse to skip politeness, please remember that there is at least one human person maintaining the code base on their free time behind most FOSS projects you profit from :wink: 

that being said, you can either downgrade to version 1.0.1 of ESP32-ENC28J60 driver, or upgrade to version 3.x.x of arduino-esp32 core

closed 2024-07-05 by tobozo