tobozo/ESP32-ENC28J60 #19: Error While Compiling

Katheesh, 2024-11-08:
|-- ESP32-ENC28J60 @ 2.0.1
|-- HTTPClient @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Compiling .pio/build/esp32_2_0_5/src/main.cpp.o
In file included from src/main.cpp:6:
.pio/libdeps/esp32_2_0_5/ESP32-ENC28J60/src/ESP32-ENC28J60.h:23:10: fatal error: Network.h: No such file or directory

tobozo, 2024-11-11:
hi,

platformio.ini should use a more recent version of espressif32 package, or an older version of this library

Katheesh, 2024-11-26:
Thank you ☺️  for your reply. 

I tried framework-arduinoespressif32@3.20017.0 and you used platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.5/esp32-2.0.5.zip.

But both are not compiled for me. 

Which version of framework-arduinoespressif32 should I use? Please advise me! Thank you ☺️

tobozo, 2024-11-26:
hi,

espressif core 2.0.5 is quite old, if you really need to work with that core you'll have to use an older version of this library, otherwise you can use pioarduino, which will pull the latest stable espressif core (probably 3.0.7).

pioarduino is like the official platformio package for espressif, but more maintained

```ini

[pioarduino-env]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
lib_deps =
  tobozo/ESP32-ENC28J60

[official-platformio-env]
platform = espressif32
platform_packages = https://github.com/espressif/arduino-esp32/releases/download/2.0.5/esp32-2.0.5.zip
lib_deps =
  tobozo/ESP32-ENC28J60 @ 1.0.1

```

Katheesh, 2024-11-27:
Thank you ☺️ man. but it's not working for me!

closed 2024-12-01 by Katheesh