tobozo/ESP32-ENC28J60 #20: pio_example

maxt, 2025-01-17:
Hi. Can't compile sample from pio_example folder:

`
Processing esp32_2_0_5 (board: esp32dev; platform: https://github.com/tasmota/platform-espressif32; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2025.1.30+sha.cadff82) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.5 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32-ENC28J60 @ 2.0.1
|-- HTTPClient @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Compiling .pio/build/esp32_2_0_5/src/main.cpp.o
Generating partitions .pio/build/esp32_2_0_5/partitions.bin
In file included from src/main.cpp:6:
.pio/libdeps/esp32_2_0_5/ESP32-ENC28J60/src/ESP32-ENC28J60.h:23:10: fatal error: Network.h: No such file or directory
`

tobozo, 2025-01-17:
hi,

either use an earlier version of this library, or use a more recent version of framework-arduinoespressif32 (3.x.x or upper)

closed 2025-01-17 by tobozo

maxt, 2025-01-17:
That is the point. 
I managed to use 1.0.1 version of library. It can be used in platformio dev environment. 
However, AFAIK there is no (3.x.x or upper) version available in platformio (https://github.com/platformio/platform-espressif32/issues/1225). 

Looks like there is no way to use 2.0.1 with platformio.

tobozo, 2025-01-17:
have you tried with [pioarduino](https://github.com/pioarduino)? unlike the official espressif core from platformio, this one is maintained and up to date 😉

maxt, 2025-01-20:
Tried the sample in pioarduino with actual versions of library and platform libraries. 
Technically, ethernet connection works,
hovewer, WiFI.onEvent(WiFiEvent);
doesn't appear to be working. 
It  doesn't catch any events in 'case' statement.
Even if I put debug print in "default:" section -it doesn't appear.

Changed while in setup() to: 
`
  while( !eth_connected) {
    Serial.print(".");
    Serial.println(ETH.localIP());
    Serial.println(ETH.gatewayIP());
    Serial.println(ETH.dnsIP());
    IPAddress local = ETH.localIP();
    if (!local.toString().equals("0.0.0.0")){
      eth_connected = true;
    }
    delay( 1000 );
  }
`
and noticed, that it's finally get parameters over DHCP and since that any network related code works.

tobozo, 2025-01-20:
good news, and thanks for sharing the solution 👍 

can you share the platformio.ini too?

possible explanation for the WiFi Events not being fired: Ethernet doesn't use WiFi Events anymore with espressif core 3.x.x, [they switched to Network Events instead](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino#L30-L58).

maxt, 2025-01-20:
My current platformio.ini
```
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/3.1.1/esp32-3.1.1.zip
board = esp32dev
framework = arduino
lib_deps = tobozo/ESP32-ENC28J60
monitor_speed = 115200
monitor_dtr = 0
monitor_rts = 0
build_flags = -DCORE_DEBUG_LEVEL=4 -DCONFIG_ETH_ENABLED=1 -DCONFIG_WIFI_ENABLED=0
```

maxt, 2025-01-20:
Can't provide PR right now, but since I got it working, so ready to post my solution:

```
#include <esp_event.h>
```

```
static void all_event_handler(void* handler_args, esp_event_base_t base, int32_t id, void* event_data)
{
    Serial.println("Caught an event!");
    Serial.println(base);
    Serial.println(id);
    
    if (ETHERNET_EVENT_CONNECTED==id){
      Serial.print("ethernet connected");
    }
    if (ETHERNET_EVENT_START==id){
      Serial.print("ethernet start");
    }

    if (IP_EVENT_ETH_GOT_IP==id && base == IP_EVENT){
      eth_connected = true;
    }
}

void setup()
{
  ESP_ERROR_CHECK(esp_event_loop_create_default());
  ESP_ERROR_CHECK(esp_event_handler_register(ESP_EVENT_ANY_BASE,ESP_EVENT_ANY_ID, all_event_handler, NULL));
  ...... 

}

```