tobozo/ESP32-ENC28J60 #21: Can't establish UDP connection using SNTP (ESP32-S3 ENC28J60) – is UDP supported?

SquareShox, 2025-03-23:
Hi,

I'm working on an ESP32-S3 project using Arduino framework and the ENC28J60 Ethernet module (via ESP32-ENC28J60 library). My PlatformIO configuration uses:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32-s3-zero]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/3.1.1/esp32-3.1.1.zip
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
upload_speed = 921600
board_build.partitions = default_16MB.csv
build_flags = 
	;-DARDUINO_USB_CDC_ON_BOOT
	-DESP32
	-D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
	;-D BUILD_WITHOUT_ETH=1
	-D LOG_LOCAL_LEVEL=5
	
lib_deps = 
	matjack1/Wiegand Protocol Library for Arduino - for esp-rfid@^1.1.1
	bblanchon/ArduinoJson@^7.2.0
	ESP32Async/ESPAsyncWebServer @ 3.6.0
	Links2004/WebSockets@^2.3.5
	tobozo/ESP32-ENC28J60

# Wykluczenie standardowej biblioteki Ethernet
lib_ignore =
	Ethernet

While HTTP and WebSocket communications work perfectly over Ethernet, I'm currently unable to connect to any NTP server using the default ESP32 SNTP library. I have tested both domain names and direct IPs, but SNTP just fails silently or times out.

All UDP ports are open on the network (verified with other devices), so it seems like a potential limitation in the library or stack.

Question:
Is UDP (specifically SNTP) supported with this Ethernet setup (ESP32-ENC28J60)?
If so, are there any known issues or required configurations to enable it?

Thanks in advance for your support!

tobozo, 2025-03-23:
hi,

thanks for your feedback 👍 

> using the default ESP32 SNTP library.

do you mean [this API](https://github.com/espressif/esp-idf/blob/master/examples/protocols/sntp/README.md)?

are you using a fixed ip or dhcp? 

I don't have an ENC28J60 to test with, however I can help with troubleshooting :

- if using dhcp, try fixed ip, else try dhcp, and confirm the problem happens in both situations
- confirm that UDP works fine when using WiFi instead of Ethernet (and specify if using dhcp or fixed ip)
- create a minimal tests sketch, no webserver, socket server, rfid or json, just the ethernet and sntp parts
- set log level to verbose
- provide the test code and console output

SquareShox, 2025-03-23:
Yes this api .
 I dont have most recent code with me right now ill test it tomorow then. 
Thank you for your help

SquareShox, 2025-03-24:
okay my bad i used wrong gateway with dhcp all works fine

closed 2025-03-24 by SquareShox